BYU-PCCL/holodeck #7: Changing resolution on Linux fails silently

wingated, 2018-01-09:
Changing: 

UrbanCity UAVAgent uav0 512 512 PrimaryPlayerCamera OrientationSensor LocationSensor VelocitySensor IMUSensor

to

UrbanCity UAVAgent uav0 480 640 PrimaryPlayerCamera OrientationSensor LocationSensor VelocitySensor IMUSensor

results in a silent hang - looks like the holodeck subprocess has died.

joshgreaves, 2018-01-10:
I recreated the bug with logging and found these lines:
[2018.01.10-01.08.25:449][  0]LogTemp: Warning: Getting buffer of size 1048576
[2018.01.10-01.08.25:449][  0]LogHolodeck: Subscribing sensor PrimaryPlayerCamera for uav0

It seems 1048576 should be height * width * sizeof(FColor).
FColor is 32 bits so that size is wrong: 480 * 640 * 32 = 9830400 so the allocated buffer is far too small.

Also, we need to have functionality to change the buffer size in Python. @mitchneutron and I discussed changing enums to individual sensor classes with member functions, instead of dictionaries from enum to value.

Handing over to @mitchneutron since the PrimaryPlayerCamera buffer is something he worked on before.

mitchneutron, 2018-01-11:
The crash due to the viewport size being different than expected should now be fixed. The fix is in the viewportBug branch of Holodeck. I'm still trying to narrow down the allocated buffer size bug.

mitchneutron, 2018-04-23:
The crash is now fixed with Nick's merged PR.

closed 2018-04-23 by mitchneutron