BYU-PCCL/holodeck #15: Roll command sign error

skyler237, 2018-01-26:
In order to produce a roll to the right, I need to give the simulation a negative roll command. Reading the state back from the orientation sensor gives a positive roll angle of equal magnitude, but opposite sign to the roll command. It seems as though there is a sign or direction error in the roll control code.

joshgreaves, 2018-01-30:
After thinking about this we've decided to put everything into line with the Unreal Axes, that is positive x is forward, positive y is right, and positive z is up.

mitchneutron, 2018-01-31:
@skyler237 To clarify, are you using the OrientationSensor or the IMUSensor? The OrientationSensor gives three vectors: the up, right, and forward vector, which can be used to calculate the exact orientation of the UAV, whereas the IMUSensor gives linear acceleration and angular velocity. Are you using one or the other to personally calculate the roll angle, or am I mistaken with something?

mitchneutron, 2018-01-31:
@skyler237 Update: I built a RotationSensor to test it, and you are right. A positive roll command yields a negative roll value. We'll be looking into it more.

joshgreaves, 2018-02-10:
@mitchneutron has this been fixed?

mitchneutron, 2018-02-11:
No it has not

mitchneutron, 2018-02-12:
This should be fixed in the next update.

closed 2018-02-12 by mitchneutron