BYU-PCCL/holodeck #16: Missing drivers?

jakelarsen17, 2018-01-27:
I've successfully installed the latest version multiple times but keep getting an error message saying that 2 drivers were not found.  The drivers are XAPOFX1_5.dll and X3DAudio1_7.dll.  When I close the error message I get an exception that says it "timed out waiting for the binary to load."  During class today one of the students mentioned that he believes there's a problem with the directories for the binary.  I don't remember the exact details, but performing a reinstall does not fix the problem.  I've tried this on each of the worlds and get the same issue.

joshgreaves, 2018-01-27:
What operating system are you running on?
When does the drivers error show up?
Are you able to load any of the worlds with holodeck.make?

Thanks

jakelarsen17, 2018-01-27:
Running windows 10. The error shows up when I run the holodeck.make
command. None of the worlds load for me.

Jake

joshgreaves, 2018-01-27:
Could you try running the Holodeck binary manually and let me know if you still get the same error?
For your convenience, just copy this into a command prompt:
%HOLODECKPATH%\worlds\WindowsDefaultWorlds\WindowsNoEditor\Holodeck\Binaries\Win64\Holodeck.exe

Also, are you running 32 bit or 64 bit Windows 10?

joshgreaves, 2018-01-27:
It seems that it may be a problem with your DirectX installation. Could you tell me your DirectX version?
Press windows button and type (and run) dxdiag. Then on the System tab you should see your DirectX version. I *think* you need at least DirectX10, but the more recent the better.

jakelarsen17, 2018-01-27:
I tried running the binary manually as described in your email and got the
same errors.  I'm on 64-bit windows 10 and have DirectX version 12.

Jake

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

joshgreaves, 2018-01-27:
When you run the binary manually it attempts to load the worlds without any Holodeck specific code. That leads me to believe that there is some incompatibility between your setup and Unreal Engine 4.

It seems those particular missing binaries are fairly common with DirectX games, there's a list of potential fixes here:
https://www.lifewire.com/fix-xapofx1-5-dll-not-found-or-missing-errors-2624318

Let me know if any of those suggestions work.

jakelarsen17, 2018-01-27:
> missing-errors-2624318

Ok reinstalling DirectX seemed to fix the problem.  But now when I run it
on any of the worlds I can't send any commands or control it.  It opens up
a display of the first image but it's like the keyboard and mouse are
completely disabled.  I have to kill the task from the task manager to get
control back.  Thanks for your help.  I really appreciate you being so
responsive, even though it's Saturday.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

joshgreaves, 2018-01-27:
No worries! Thanks for testing it out. That's normal on Windows. The way around it is to alt tab out of it, the game won't tick until you send a step command to the environment. We will look into ways to make that part more user friendly.

jakelarsen17, 2018-01-27:
Ok got it.  Thank you!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2018-01-27 by joshgreaves