BYU-PCCL/holodeck #26: Holodeck Worlds

pmarke, 2018-02-04:
I installed the latest version of Holodeck, and I am no longer able to load any worlds. They simply are not found.

mitchneutron, 2018-02-06:
Hey I’ve been looking to narrow down why this is happening. Do you get any errors thrown by the install script or does it state that the install was successful?

pmarke, 2018-02-06:
It says that it was successful

mitchneutron, 2018-02-07:
Okay, so I think I found what was going on. It was definitely a problem on our end. If you try it again then it should work correctly. The updated code expected a json file for configurations, but the packaged world didn't have that json file. We've now updated both the code and the worlds.

mitchneutron, 2018-02-08:
I'm marking this as closed as the bug should be gone.

closed 2018-02-08 by mitchneutron