BYU-PCCL/holodeck #70: Can't pip install on ubuntu 16.04

wingated, 2018-08-10:
It thinks I'm on windows.  Error log below:

``wingated@coban:~$ sudo -H pip3 install holodeck
Collecting holodeck
  Downloading https://files.pythonhosted.org/packages/28/b4/471bf45fe6f723f4753d50379f6b402b319a8c142aafb8f0700f33c78054/holodeck-0.0.21.tar.gz
  Ignoring pywin32: markers 'platform_system == "Windows"' don't match your environment
Collecting posix_ipc>=1.0.0; platform_system == "Linux" (from holodeck)
  Downloading https://files.pythonhosted.org/packages/c9/3e/54217da71aa26b488295d878df4d3132093253b4ae5798ac66fcb6921ef0/posix_ipc-1.0.4.tar.gz (78kB)
    100% |████████████████████████████████| 81kB 2.8MB/s 
Building wheels for collected packages: holodeck, posix-ipc
  Running setup.py bdist_wheel for holodeck ... done
  Stored in directory: /root/.cache/pip/wheels/c1/27/8b/2dbcb4c5343f796f32eb48f5e07d6b4612b7462d05cfb5b4fa
  Running setup.py bdist_wheel for posix-ipc ... done
  Stored in directory: /root/.cache/pip/wheels/5a/5c/17/d0e2d421abaf3b4097bf9db11108380d734195c6d15c24269d
Successfully built holodeck posix-ipc
**holodeck 0.0.21 requires pywin32>=1.0, which is not installed.**
Installing collected packages: posix-ipc, holodeck
Successfully installed holodeck-0.0.21 posix-ipc-1.0.4

joshgreaves, 2018-08-10:
Update: It actually successfully installed (see last line) but logged an error message for requiring pywin32. This error message shouldn't show up.

wingated, 2018-08-16:
It appears to be a pip bug.  I'm not sure which versions are affected, but pip installing on stock 16.04 doesn't work; after upgrading pip to the latest version (18.0 as of this writing), it installs but with an error message.

closed 2018-10-08 by joshgreaves