BYU-PCCL/holodeck #72: Cyberpunk city install errors

wingated, 2018-08-16:
While installing Cyberpunkcity, several things went wrong:

1) Some sort of unicode error?  It only affected display; the world still downloaded in the background

2) Once downloaded, 

'''holodeck.install('CyberPunkCity')
Installing CyberPunkCity at /home/wingated/.local/share/holodeck
File size: 1.18 GB
|                    | 4%Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.5/dist-packages/holodeck/packagemanager.py", line 147, in file_writer_worker
    sys.stdout.write("\r|" + blocks + spaces + "| %d%%" % int_percent)
UnicodeEncodeError: 'ascii' codec can't encode character '\u2589' in position 2: ordinal not in range(128)

Unpacking worlds...
---------------------------------------------------------------------------
BadZipFile                                Traceback (most recent call last)
<ipython-input-19-5304af1a0af3> in <module>()
----> 1 holodeck.install('CyberPunkCity')

/usr/local/lib/python3.5/dist-packages/holodeck/packagemanager.py in install(package_name)
     97     install_path = os.path.join(holodeck_path, "worlds")
     98     binary_url = binary_website + util.get_os_key() + "_" + package_url
---> 99     _download_binary(binary_url, install_path)
    100     if os.name == "posix":
    101         _make_binary_excecutable(package_name, install_path)

/usr/local/lib/python3.5/dist-packages/holodeck/packagemanager.py in _download_binary(binary_location, worlds_path, block_size)
    164     # Unzip the binary
    165     print("Unpacking worlds...")
--> 166     with zipfile.ZipFile(tmp_fd, 'r') as zip_file:
    167         zip_file.extractall(worlds_path)
    168 

/usr/lib/python3.5/zipfile.py in __init__(self, file, mode, compression, allowZip64)
   1024         try:
   1025             if mode == 'r':
-> 1026                 self._RealGetContents()
   1027             elif mode in ('w', 'x'):
   1028                 # set the modified flag so central directory gets written

/usr/lib/python3.5/zipfile.py in _RealGetContents(self)
   1091             raise BadZipFile("File is not a zip file")
   1092         if not endrec:
-> 1093             raise BadZipFile("File is not a zip file")
   1094         if self.debug > 1:
   1095             print(endrec)

BadZipFile: File is not a zip file
'''

wingated, 2018-08-16:
I should add: I manually downloaded the zip file, unzipped it, changed the Holodeck binary to be executable, and it worked fine...

MaxRobinsonTheGreat, 2018-09-11:
Should be fixed but needs to be tested.

closed 2018-10-08 by joshgreaves