BYU-PCCL/holodeck #82: Can't control "nav0" in "CyberPunkCity"

joshgreaves, 2018-09-04:


nickwalton, 2018-09-07:
I wonder if this is a post process volume issue?

nickwalton, 2018-09-07:
I tested this and found that the nav agent is able to move. Perhaps the issue was giving him coordinates that he is unable to move to? I was able to get him to move to near the uav's position using this code: 
` env = holodeck.make("CyberPunkCity")

    cmd0 = np.array([0, 0, 10, 10])
    cmd1 = np.array([-6000,-3600,0])

    env.set_control_scheme("uav0",1)
    for i in range(10):
        env.reset()
        # This will queue up a new agent to spawn into the environment, given that the coordinates are not blocked.
        sensors = [Sensors.PIXEL_CAMERA, Sensors.LOCATION_SENSOR, Sensors.VELOCITY_SENSOR]
        env.tick()  # Tick the environment once so the second agent spawns before we try to interact with it.

        env.act("uav0", cmd0)
        env.act("nav0", cmd1)
        for _ in range(5000):
            states = env.tick()
            uav0_terminal = states["uav0"][Sensors.TERMINAL]`

closed 2018-09-07 by nickwalton