BYU-PCCL/holodeck #83: Can't control "sphere0" in "MazeWorld"

joshgreaves, 2018-09-04:


joshgreaves, 2018-09-06:
Here is my code, running on develop branch with the latest world downloaded:
```
import holodeck
from holodeck.sensors import Sensors
env = holodeck.make("MazeWorld")
for i in range(120):
    s, _, _, _ = env.step(0)
    print(i, s[Sensors.LOCATION_SENSOR])
```
result:
```
0 [0. 0. 0.]
1 [0. 0. 0.]
2 [0. 0. 0.]
3 [0. 0. 0.]
4 [0. 0. 0.]
5 [0. 0. 0.]
6 [0. 0. 0.]
7 [0. 0. 0.]
8 [0. 0. 0.]
9 [0. 0. 0.]
10 [0. 0. 0.]
11 [0. 0. 0.]
12 [0. 0. 0.]
13 [0. 0. 0.]
14 [0. 0. 0.]
15 [0. 0. 0.]
16 [0. 0. 0.]
17 [0. 0. 0.]
18 [0. 0. 0.]
19 [0. 0. 0.]
20 [0. 0. 0.]
21 [0. 0. 0.]
22 [0. 0. 0.]
23 [0. 0. 0.]
24 [0. 0. 0.]
25 [0. 0. 0.]
26 [0. 0. 0.]
27 [0. 0. 0.]
28 [0. 0. 0.]
29 [0. 0. 0.]
30 [0. 0. 0.]
31 [0. 0. 0.]
32 [0. 0. 0.]
33 [0. 0. 0.]
34 [0. 0. 0.]
35 [0. 0. 0.]
36 [0. 0. 0.]
37 [0. 0. 0.]
38 [0. 0. 0.]
39 [0. 0. 0.]
40 [0. 0. 0.]
41 [0. 0. 0.]
42 [0. 0. 0.]
43 [0. 0. 0.]
44 [0. 0. 0.]
45 [0. 0. 0.]
46 [0. 0. 0.]
47 [0. 0. 0.]
48 [0. 0. 0.]
49 [0. 0. 0.]
50 [0. 0. 0.]
51 [0. 0. 0.]
52 [0. 0. 0.]
53 [0. 0. 0.]
54 [0. 0. 0.]
55 [0. 0. 0.]
56 [0. 0. 0.]
57 [0. 0. 0.]
58 [0. 0. 0.]
59 [0. 0. 0.]
60 [0. 0. 0.]
61 [0. 0. 0.]
62 [0. 0. 0.]
63 [0. 0. 0.]
64 [0. 0. 0.]
65 [0. 0. 0.]
66 [0. 0. 0.]
67 [0. 0. 0.]
68 [0. 0. 0.]
69 [0. 0. 0.]
70 [0. 0. 0.]
71 [0. 0. 0.]
72 [0. 0. 0.]
73 [0. 0. 0.]
74 [0. 0. 0.]
75 [0. 0. 0.]
76 [0. 0. 0.]
77 [0. 0. 0.]
78 [0. 0. 0.]
79 [0. 0. 0.]
80 [0. 0. 0.]
81 [0. 0. 0.]
82 [0. 0. 0.]
83 [0. 0. 0.]
84 [0. 0. 0.]
85 [0. 0. 0.]
86 [0. 0. 0.]
87 [0. 0. 0.]
88 [0. 0. 0.]
89 [0. 0. 0.]
90 [0. 0. 0.]
91 [0. 0. 0.]
92 [0. 0. 0.]
93 [0. 0. 0.]
94 [0. 0. 0.]
95 [0. 0. 0.]
96 [0. 0. 0.]
97 [0. 0. 0.]
98 [0. 0. 0.]
99 [0. 0. 0.]
100 [0. 0. 0.]
101 [0. 0. 0.]
102 [0. 0. 0.]
103 [0. 0. 0.]
104 [0. 0. 0.]
105 [0. 0. 0.]
106 [0. 0. 0.]
107 [0. 0. 0.]
108 [0. 0. 0.]
109 [0. 0. 0.]
110 [0. 0. 0.]
111 [0. 0. 0.]
112 [0. 0. 0.]
113 [0. 0. 0.]
114 [0. 0. 0.]
115 [0. 0. 0.]
116 [0. 0. 0.]
117 [0. 0. 0.]
118 [0. 0. 0.]
119 [0. 0. 0.]
```

nickwalton, 2018-09-07:
Turns out I was testing it with the old packaged worlds. I was able to reproduce it with the new packaged worlds and fixed it by changing 

"unsigned int GetRawActionSizeInBytes() const override { return 2; };"
in SphereRobot.h to
"unsigned int GetRawActionSizeInBytes() const override { return 2 * sizeof(float); };"

closed 2018-09-07 by nickwalton