BYU-PCCL/holodeck #108: Tasks in the environment

ankeshanand, 2018-10-05:
Hi,

Great work, excited to try out this environment. I was wondering what sort of tasks are integrated with each environment, since I couldn't find the documentation around it?

(x-posted from https://github.com/BYU-PCCL/holodeck-engine/issues/104, believe this is the more recent repo)

MaxRobinsonTheGreat, 2018-10-05:
@ankeshanand I have written some basic documentation of the tasks for the environments in DefaultWorlds. Take a look at the [tasks](https://github.com/BYU-PCCL/holodeck/blob/master/docs/tasks.md) and [worlds](https://github.com/BYU-PCCL/holodeck/blob/master/docs/worlds.md) docs to get some direction. Thanks for the input! Let me know if anything is lacking

ankeshanand, 2018-10-06:
Thanks @MaxRobinsonTheGreat, this is def helpful! Could you also point me to the code where these reward functions are specified? I did a quick lookup but couldn't find it.

nickwalton, 2018-10-09:
@ankeshanand The code for these reward functions is currently on the blueprint side which can only be edited/viewed when editing the Unreal project with Unreal editor. These can be viewed and edited by cloning the unreal project repo (https://github.com/byu-pccl/holodeck-engine) and editing it with Unreal Editor. However we realize these should be more transparent and are currently planning on moving them over to the C++ code in the near future.

closed 2018-10-09 by ankeshanand