BYU-PCCL/holodeck #113: Clockspeed

tvturnhout, 2018-10-09:
How does Holodeck compare to for example Airsim when it comes to performance, resource usage and increasing simulation clockspeed while maintaining stable simulation?

Great project, highly support it!

joshgreaves, 2018-10-10:
@nickwalton will do some benchmarking and get back on this.

tvturnhout, 2018-10-11:
Great, looking forward to it!

nickwalton, 2018-10-16:
I did a cursory comparison of airsim and holodeck in terms of rendering and image capture.  
Capturing 30 images per second I got roughly 20 - 30 FPS on airsim 
Doing the same on holodeck I got 45-55 FPS. 

When it comes to simulation clockspeed I'm not quite sure how airsim works with increasing the clockspeed and the performance/stability effects. With holodeck the default is set at 30 physics steps and frame renders per unreal second. But the simulation will go at the fastest speed it can, which is usually slightly below 2 unreal seconds for every real world second. Decreasing the physics and rendering steps per second will increase this ratio.

Holodeck uses shared memory to communicate between the python api and unreal executable enabling faster inter process communication than TCP/IP which is what Airsim uses.

tvturnhout, 2018-10-26:
Thanks for the info, the x2 rule of thumb is very useful. My findings were that Holodeck can maintain stability with noticeably faster clockspeeds than Airsim :)!

closed 2019-01-11 by nickwalton