BYU-PCCL/holodeck #114: Environment Make, Semaphore Busy

emcyroyale, 2018-10-10:
When I try to make an environment, I receive the below error. I'm running this in Ubuntu 16.04.5.
How can I fix or debug this issue?

### Code
env = holodeck.make("UrbanCity")

### Error
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/holodeck/environments.py", line 385, in __linux_start_process__
    loading_semaphore.acquire(100)
posix_ipc.BusyError: Semaphore is busy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/holodeck/holodeck.py", line 54, in make
    return HolodeckEnvironment(**param_dict)
  File "/usr/local/lib/python3.5/dist-packages/holodeck/environments.py", line 102, in __init__
    self.__linux_start_process__(binary_path, task_key, gl_version, verbose=verbose)
  File "/usr/local/lib/python3.5/dist-packages/holodeck/environments.py", line 387, in __linux_start_process__
    raise HolodeckException("Timed out waiting for binary to load")
holodeck.exceptions.HolodeckException: Timed out waiting for binary to load

`

nickwalton, 2018-10-10:
Are you running holodeck as sudo? Because sometimes that issue happens when sudo is used because unreal executables don't allow themselves to be run with root privileges.

emcyroyale, 2018-10-11:
Thanks for responding. I did not use sudo to run anything.

joshgreaves, 2018-10-11:
Did you install using docker (as opposed to pip)? If so, did you make sure to create another user and sign in before installing and running environments? The environment won't load if you run with root permissions or as the root user.

Otherwise, please could you run `env = holodeck.make("UrbanCity", verbose=True)` and print the output here.

emcyroyale, 2018-10-11:
I am using docker (I.e docker run -it pccl/holodeck:ubuntu16.04-without-worlds) and I switched to the holodeckuser account.

>>> env = hd.make("UrbanCity", verbose=True)

Increasing per-process limit of core file size to infinity.
- Existing per-process limit (soft=18446744073709551615, hard=18446744073709551615) is enough for us (need only 18446744073709551615)
sh: 1: xdg-user-dir: not found
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(-917496897) SRandInit(-917496897).
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.046645
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
LogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
LogInit: Build: ++UE4+Release-4.17-CL-0
LogInit: Engine Version: 4.17.2-0+++UE4+Release-4.17
LogInit: Compatible Engine Version: 4.17.0-0+++UE4+Release-4.17
LogInit: Net CL: 0
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 2
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 20
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 28
LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 0
LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0
LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0
LogDevObjectVersion:   Dev-Physics (78F01B33-EBEA-4F98-B9B4-84EACCB95AA2): 0
LogDevObjectVersion:   Dev-Platform (6631380F-2D4D-43E0-8009-CF276956A95A): 0
[... 90 lines cut ...]
[Callstack]  13  0x000000000393fd1d  RHIInit(bool) [C:\Users\Max\Documents\UnrealEngine\Engine\Source\Runtime\RHI\Private/DynamicRHI.cpp, line 178]
[Callstack]  14  0x00000000022ffa54  FEngineLoop::PreInit(wchar_t const*) [C:\Users\Max\Documents\UnrealEngine\Engine\Source\Runtime\Launch\Private/LaunchEngineLoop.cpp, line 1713]
[Callstack]  15  0x000000000230aff8  GuardedMain(wchar_t const*) [C:\Users\Max\Documents\UnrealEngine\Engine\Source\Runtime\Launch\Private/Launch.cpp, line 42]
[Callstack]  16  0x00000000057c2274  CommonLinuxMain(int, char**, int (*)(wchar_t const*)) [C:\Users\Max\Documents\UnrealEngine\Engine\Source\Runtime\Linux\LinuxCommonStartup\Private/LinuxCommonStartup.cpp, line 236]
[Callstack]  17  0x00007fdffb682830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fdffb682830]
[Callstack]  18  0x00000000022fce3e  /home/holodeckuser/.local/share/holodeck/worlds/LinuxDefaultWorlds/LinuxNoEditor/Holodeck/Binaries/Linux/Holodeck() [0x22fce3e]

[2018.10.11-17.11.43:693][  0]LogExit: Executing StaticShutdownAfterError
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/holodeck/environments.py", line 384, in __linux_start_process__
    loading_semaphore.acquire(100)
posix_ipc.BusyError: Semaphore is busy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... cut ...]
  File "/usr/local/lib/python3.5/dist-packages/holodeck/environments.py", line 101, in __init__
    self.__linux_start_process__(binary_path, task_key, gl_version, verbose=verbose)
  File "/usr/local/lib/python3.5/dist-packages/holodeck/environments.py", line 386, in __linux_start_process__
    raise HolodeckException("Timed out waiting for binary to load. Ensure that holodeck is not being run with root priveleges.")
holodeck.exceptions.HolodeckException: Timed out waiting for binary to load. Ensure that holodeck is not being run with root priveleges.

nickwalton, 2018-10-11:
Hi @emcyroyale I was able to reproduce and fix your issue. The documentation we had on our README said to run "docker run -it pccl/holodeck:ubuntu16.04-without-worlds" but that was actually incorrect. The proper command is "**nvidia-docker** run -it pccl/holodeck:ubuntu16.04-without-worlds." To render with GPU accelerated rendering, the docker container must access cuda and GPU resources which it can only do using nvidia docker. I've updated the README to have the proper command. Thank you for making us aware of that issue!

closed 2018-10-12 by emcyroyale

Toga199915, 2024-03-20:
Hello I have the same problem, could you help me please? I am not using docker.
I have this in the console:
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_sampler_mirror_clamp_to_edge
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_separate_depth_stencil_layouts
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_shader_atomic_int64
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_shader_draw_parameters
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_storage_buffer_storage_class
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_swapchain
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_uniform_buffer_standard_layout
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: -    Found device extension VK_KHR_variable_pointers
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: Using Device 0: Geometry 1 Tessellation 1 BufferAtomic64 1
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: Found 1 Queue Families
[2024.03.20-20.10.35:020][  0]LogVulkanRHI: Display: Initializing Queue Family 0: 1 queues Gfx Compute Xfer
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: Using device layers
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: Using device extensions
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_driver_properties
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_get_memory_requirements2
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_dedicated_allocation
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_swapchain
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_maintenance1
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_maintenance2
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_EXT_scalar_block_layout
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_shader_atomic_int64
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: * VK_KHR_image_format_list
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: 1 Device Memory Heaps; Max memory allocations -1
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display:  0: Flags 0x1 Size 2147483648 (2048.00 MB) GPU
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: 1 Device Memory Types (Not unified)
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display:  0: Flags 0x0000f Heap  0  Local HostVisible HostCoherent HostCached     
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: Host Allocation Percentage   0.00% -          0.00MB /    0.000MB
[2024.03.20-20.10.35:025][  0]LogVulkanRHI: Display: Memory Budget unavailable
[2024.03.20-20.10.35:026][  0]LogVulkanRHI: Display: FVulkanPipelineStateCacheManager: Binary pipeline cache '../../../Holodeck/Saved/VulkanPSO.cache.10005.0' not found.
[2024.03.20-20.10.35:030][  0]LogRHI: Texture pool is 1433 MB (70% of 2048 MB)
[2024.03.20-20.10.35:030][  0]LogRendererCore: Ray tracing is disabled. Reason: r.RayTracing=0.
[2024.03.20-20.10.35:041][  0]LogShaderLibrary: Display: Using ../../../Holodeck/Content/ShaderArchive-Global-SF_VULKAN_SM5.ushaderbytecode for material shader code. Total 1859 unique shaders.
[2024.03.20-20.10.35:041][  0]LogShaderLibrary: Display: Cooked Context: Using Shared Shader Library Global
[2024.03.20-20.10.35:041][  0]LogShaderLibrary: Display: Logical shader library 'Global' has been created, components 1
[2024.03.20-20.10.35:042][  0]LogTemp: Warning: Clearing the OS Cache
[2024.03.20-20.10.35:042][  0]LogODSC: ODSC Thread active.
[2024.03.20-20.10.35:044][  0]LogPakFile: New pak file ../../../Holodeck/Content/Paks/Holodeck-LinuxNoEditor.pak added to pak precacher.
[2024.03.20-20.10.35:088][  0]LogSlate: Using FreeType 2.10.0
[2024.03.20-20.10.35:089][  0]LogSlate: SlateFontServices - WITH_FREETYPE: 1, WITH_HARFBUZZ: 1
[2024.03.20-20.10.35:138][  0]LogShaderLibrary: Display: Using ../../../Holodeck/Content/ShaderArchive-Holodeck-SF_VULKAN_SM5.ushaderbytecode for material shader code. Total 5831 unique shaders.
[2024.03.20-20.10.35:138][  0]LogShaderLibrary: Display: Cooked Context: Using Shared Shader Library Holodeck
[2024.03.20-20.10.35:138][  0]LogShaderLibrary: Display: Logical shader library 'Holodeck' has been created, components 1
[2024.03.20-20.10.35:138][  0]LogRHI: Display: Opened pipeline cache after state change and enqueued 0 of 0 tasks for precompile.
[2024.03.20-20.10.35:140][  0]LogShaderLibrary: Display: Using ../../../Holodeck/Content/ShaderArchive-Holodeck-SF_VULKAN_SM5.ushaderbytecode for material shader code. Total 5831 unique shaders.
[2024.03.20-20.10.35:140][  0]LogShaderLibrary: Display: Cooked Context: Using Shared Shader Library Holodeck
[2024.03.20-20.10.35:140][  0]LogRHI: Display: Opened pipeline cache after state change and enqueued 0 of 0 tasks for precompile.
[2024.03.20-20.10.35:140][  0]LogRHI: Display: Failed to open default shader pipeline cache for Holodeck using shader platform 20.
[2024.03.20-20.10.35:140][  0]LogInit: Using OS detected language (en-US).
[2024.03.20-20.10.35:140][  0]LogInit: Using OS detected locale (en-US).
[2024.03.20-20.10.35:140][  0]LogTextLocalizationManager: No localization for 'en-US' exists, so 'en' will be used for the language.
[... cut ...]
    return HoloOceanEnvironment(**param_dict)
  File "/home/jason/.local/lib/python3.8/site-packages/holoocean/environments.py", line 193, in __init__
    self._client.acquire(self._timeout)
  File "/home/jason/.local/lib/python3.8/site-packages/holoocean/holooceanclient.py", line 91, in acquire
    self._get_semaphore_fn(self._semaphore2, timeout)
  File "/home/jason/.local/lib/python3.8/site-packages/holoocean/holooceanclient.py", line 72, in posix_acquire_semaphore
    sem.acquire(timeout)
posix_ipc.BusyError: Semaphore is busy