BYU-PCCL/holodeck #115: mmap length is greater than file size

joshgreaves, 2018-10-10:
On `holodeck.make('EuropeanForest')`, world loaded but this error was logged:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/holodeck/holodeck.py", line 54, in make
    return HolodeckEnvironment(**param_dict)
  File "/usr/local/lib/python3.6/dist-packages/holodeck/environments.py", line 113, in __init__
    self._add_agents(agent_definitions)
  File "/usr/local/lib/python3.6/dist-packages/holodeck/environments.py", line 445, in _add_agents
    prepared_agents = self._prepare_agents(agent_definitions)
  File "/usr/local/lib/python3.6/dist-packages/holodeck/environments.py", line 433, in _prepare_agents
    return [self._prepare_agents(x)[0] for x in agent_definitions]
  File "/usr/local/lib/python3.6/dist-packages/holodeck/environments.py", line 433, in <listcomp>
    return [self._prepare_agents(x)[0] for x in agent_definitions]
  File "/usr/local/lib/python3.6/dist-packages/holodeck/environments.py", line 434, in _prepare_agents
    return [agent_definitions.type(client=self._client, name=agent_definitions.name)]
  File "/usr/local/lib/python3.6/dist-packages/holodeck/agents.py", line 54, in __init__
    self._teleport_bool_buffer = self._client.malloc(name + "_teleport_flag", [1], np.uint8)
  File "/usr/local/lib/python3.6/dist-packages/holodeck/holodeckclient.py", line 100, in malloc
    self._memory[key] = Shmem(key, shape, dtype, self._uuid)
  File "/usr/local/lib/python3.6/dist-packages/holodeck/shmem.py", line 44, in __init__
    self._mem_pointer = mmap.mmap(f, size_bytes)
ValueError: mmap length is greater than file size
```

nickwalton, 2019-01-11:
Finished in latest phase

closed 2019-01-11 by nickwalton

jaydenmilne, 2019-06-21:
I just ran into this on `0.2.2dev` - might of had a regression

commit 689ad57b (2018-12-19): Merge pull request #155 from BYU-PCCL/fix-mmap-length-issue

Fix #115 mmap length greater than file size
commit 763c4512 (2018-12-19): Fix #115 mmap length greater than file size

Previously there was a race condition where after `os.ftruncate()`
the size of the new file wouldn't be updated by the time `mmap.mmap()`
was called. An fsync seems to be the solution[1].

Also resolved an issue where open files were not closed. Not all of the
file leaks have been resolved, there is a TODO describing the problem better.

[1] https://lwn.net/Articles/457667/ see "When Should You Fsync?"