BYU-PCCL/holodeck #176: Ability to auto-tile windows

joshgreaves, 2019-02-11:
When I want to watch my training during debugging I have to manually slide each window apart because they start in the same place. It would be great if they could tile automatically.

Also, it would be good if I could specify the x, y coordinate of the first window.

jaydenmilne, 2019-05-20:
This is probably out of the scope of this project. [There are window tiling utilities available for most operating systems](https://www.google.com/search?client=ubuntu&channel=fs&q=tile+group+of+windows+linux&ie=utf-8&oe=utf-8)

jaydenmilne, 2019-06-04:
Just kidding, UnrealEngine appears to have a feature for this:

https://answers.unrealengine.com/questions/700447/cant-specify-custom-window-positions-in-windowed-m.html

We'd have to think what we want this feature to look like. I don't think its feasible to have the environments communicate with each other, we could leave it up to the user.

In `holodeck.make` we could let the user specify a poistion in a grid `(0,2)` and start coordinates `(400,400)` and it figures out where `(0,2)` is based off the width/height of the window and the start offset.