BYU-PCCL/holodeck #178: set_state and teleport functions only set location (develop branch)

sethmnielsen, 2019-03-07:
Note: I'm working on the `develop` branch.

For some reason, only location seems to be getting set correctly when using the `set_state` function. I'm using the UAV agent. 

The other state values (rotation, velocity, and angular velocity) don't seem to be affected at all when calling the function. I've stepped through the code and it seems that the state values are carried all the way down to `HolodeckEnvironment.tick()`, but only the `LocationSensor` (index 9 in `_sensor_map`) is returned from `self._get_full_state()` with a correctly modified value. 

I get the same behavior when using `teleport` as well. Also, are both the `RotationSensor` and `OrientationSensor` working correctly for the UAV?

@nickwalton

nickwalton, 2019-03-07:
Are you on the latest develop? If so that might be the problem. If not what
is the last commit or when was it last pulled?

sethmnielsen, 2019-03-07:
No, I'm on commit 81bb2291fcf2718b8d6eb8d09da903a8e4e7fae8.

Also, I should mention that I'm working off a fork where I merged the `BoatAgent` branch into `develop`.

sethmnielsen, 2019-03-07:
More info: I'm working in the Ocean world, but I tried switching to UrbanCity and had the same issue.

I also don't see anything that the code from `BoatAgent` could have done to cause this. Here's a link to a PR I made on my fork just in case you wanted to see the changes I've made since 81bb229 (it's only the BoatAgent merge and changes to the example files): https://github.com/sethmnielsen/holodeck/pull/3

sethmnielsen, 2019-03-14:
@nickwalton any updates on this issue?

I noticed that on master, `teleport` does correctly set the orientation. However, on either the BoatAgent or develop branches, `teleport` is broken and has no effect on the agent's orientation (but location is set just fine).

Also, somewhat related to this - is there any way the UAV agent can be given a RotationSensor? It would be really convenient to know the drone's yaw, pitch and roll. Adding the sensor to Holodeck's `config.json` only creates another field in `state` that always reads `[0, 0, 0]`.

nickwalton, 2019-03-18:
Hey sorry I didn't get it done before I left on my cruise but I'll work to
try it done right away now that I'm back

nickwalton, 2019-03-19:
Hey so I tested this issue and confirmed it to be an issue with the
packaged ocean binary. The set_state commands work in develop, but not when
I tried it with the ocean world. I'll talk to max about getting it
repackaged, but it might make more sense to just wait till we release again
in a couple weeks. Though I'm not sure if you would need it before then.

sethmnielsen, 2019-03-21:
Okay, got it. It is a bit urgent. If there is any way it could be fixed before then, that would be greatly appreciated. I emailed @MaxRobinsonTheGreat about it as well.

nickwalton, 2019-03-21:
Ok good to know. We'll try and asap it. We just got automated building
working for the default worlds so I'll try and get ocean world in that.

jaydenmilne, 2019-08-01:
I believe this should be fixed for at least the `teleport` command - we've validated it with the  `test_rotation_sensor_after_teleport` test.

We don't have any tests for `set_physics_state` (renamed from `set_state`) yet.

jaydenmilne, 2019-08-01:
> is there any way the UAV agent can be given a RotationSensor? It would be really convenient to know the drone's yaw, pitch and roll. Adding the sensor to Holodeck's `config.json` only creates another field in `state` that always reads `[0, 0, 0]`.

@sethmnielsen I believe this is because you were using an older version of Holodeck that couldn't dynamically spawn sensors and agents. We've implemented that now in Scenario files (see https://holodeck.readthedocs.io/en/latest/packages/docs/scenarios.html)

daniekpo, 2020-06-13:
@jaydenmilne from the conversation it looks like this issue is resolved. What needs to be done here?

jaydenmilne, 2020-08-20:
@daniekpo Missed this email, sorry. This is open because we had no tests for `set_physics_state`

mattpow35, 2020-11-18:
Added tests #436

closed 2020-11-18 by mattpow35