BYU-PCCL/holodeck #181: Issue with deprecated agent

joshgreaves, 2019-03-25:
The `_get_worlds_map` function causes an error when I try to load a custom built world using develop branch code. 

Error:
```
Traceback (most recent call last):
  File "/home/josh/Documents/cogarch/ltp.py", line 13, in <module>
    main()
  File "/home/josh/Documents/cogarch/ltp.py", line 6, in main
    env = holodeck.make('AndroidPlaytime')
  File "/home/josh/Documents/cogarch/holodeck/holodeck.py", line 40, in make
    holodeck_worlds = _get_worlds_map()
  File "/home/josh/Documents/cogarch/holodeck/holodeck.py", line 69, in _get_worlds_map
    "agent_definitions": [AgentDefinition(**x) for x in level["agents"]],
  File "/home/josh/Documents/cogarch/holodeck/holodeck.py", line 69, in <listcomp>
    "agent_definitions": [AgentDefinition(**x) for x in level["agents"]],
  File "/home/josh/Documents/cogarch/holodeck/agents.py", line 404, in __init__
    self.type = AgentDefinition._type_keys[agent_type] if isinstance(agent_type, str) else agent_type
KeyError: 'DiscreteSphereAgent'
```

It looks like the problem is that I have an old version of `DefaultWorlds` installed:
LinuxDefaultWorlds/config.json
```
{  
   "name":"DefaultWorlds",
   "platform":"Linux",
   "version":1.04,
   "path" : "LinuxNoEditor/Holodeck/Binaries/Linux/Holodeck",
   "maps":[  
      {  
         "name":"MazeWorld",
         "agents":[  
            {  
               "agent_name":"sphere0",
               "agent_type":"DiscreteSphereAgent",
```
and the DiscreteSphereAgent has been deprecated in the latest code.
It would be good if this was backward compatible, and used a warning.
I guess this would be something that could be fixed by a revamped package manager that can manage different versions of different worlds with different versions of Holodeck.

jaydenmilne, 2019-03-25:
We have that feature (managing different versions of binaries) planned.

But yeah, this should be a simple(ish) fix

jaydenmilne, 2019-05-20:
This should be a non-issue now that Holodeck binaries/configs are copied into a version specific folder

closed 2019-05-20 by jaydenmilne