BYU-PCCL/holodeck #183: Can't set camera height/width from code

joshgreaves, 2019-03-25:
I'm trying to set the camera for the android agent to be 224x224.
In `HolodeckEnvironment.__linux_start_process__` I print `self._camera_height` and `self._camera_width` and it is correct, but when I get data from the RGBCamera it is shape `256, 256, 4`. It looks like the command line argument no longer works.

Also, it would be convenient to truncate the `a` from the `rgba` result of the `RGBCamera`, so that it just returns `(256, 256, 3)`. The alpha value is always the same, and I don't see why anyone would need it.

jaydenmilne, 2019-05-20:
Fixed in #232

jaydenmilne, 2019-05-20:
BTW it seems Airsim stopped copying out the alpha channel and called it a "performance improvement" - might be good to look into

closed 2019-05-20 by jaydenmilne