BYU-PCCL/holodeck #196: `HolodeckEnvironment` class is huge

jaydenmilne, 2019-05-01:
1. Trim down constructor arguments
2. Move debug drawing functions somewhere else
    - Subclass? ie `env.debug.drawline()`
    - takes environment as variable? ie `drawDebugLine(env)`
3. Move weather/time functions somewhere else

Should the environment be an object acted upon more?

jaydenmilne, 2019-05-02:
There are a lot of commands that are simple wrappers for `Command` objects - maybe instead of providing these helper functions, we can just educate the user into sending their own commands.

jaydenmilne, 2019-05-06:
See #207

jaydenmilne, 2019-08-01:
I've trimmed down the constructor arguments. I've also removed a lot of the agent manipulation stuff (teleport, set physics state, etc) to the Agent class as well, we just need to move the debug stuff and the weather stuff.

jaydenmilne, 2020-01-09:
When  #263 is done this can be closed probably

vinhowe, 2020-03-11:
@jaydenmilne Can this be closed now that #372 is in?

PR #415 (2020-06-16): Move debug draw commands from HolodeckEnvironment to util
An attempt to further slim down `HolodeckEnvironment` so we can close #196.
closed 2021-01-29 by vinhowe