BYU-PCCL/holodeck #252: Significant discrepancy between Viewport and RGBCamera

MaxRobinsonTheGreat, 2019-05-17:
**Describe**
There can be major differences between the viewport and rgbcamera, which can cause a misunderstanding in what the user thinks the agent is seeing. This is most apparent in the InfiniteForest world, where the lighting is almost pitch black in the camera sensor but looks normal from the viewport, well after the pre-start steps. 

**To Reproduce**
Run the infiniteforest world and display the rgbcamera sensor's captured pixels. Most are black, and only a few are slightly lighter than that. It is not a broken sensor - an image of the world is being captured but is extremely dark.

**Screenshots**
RGBCamera capture after 200 ticks:
[image]

Similar capture from the viewport:
[image]

**Version Information:**
 - OS: Linux and Windows
 - Version 0.2.1
 - InfiniteForest-MaxDistance

We may fix this by making the infinite forest world brighter, but this would cause the world to look washed out in the viewport and fails to fix the root of the problem.

jaydenmilne, 2019-05-18:
Are we sure this isn't present in 0.1.0?

jaydenmilne, 2019-05-20:
This looks helpful: http://www.cs.jhu.edu/~qiuwch/unrealcv/plugin/3de2bd7/Source/UnrealCV/Private/GTCaptureComponent.cpp

jaydenmilne, 2019-05-20:
Reproduced on v0.1.0
[image]
[image]

jaydenmilne, 2019-05-24:
This looks like an HDR thing to me

jaydenmilne, 2019-06-05:
Or it could be because of the small resolution of the camera

MaxRobinsonTheGreat, 2019-06-20:
Increased lighting in infinite forest did not fix the issue. This did: BYU-PCCL/holodeck-engine#180

closed 2019-06-20 by MaxRobinsonTheGreat