BYU-PCCL/holodeck #262: Fail task when agents are abused

jaydenmilne, 2019-05-21:
**Describe the new feature you'd like**
Currently, tasks will not go terminal if agents have crashed or been damaged, ie if the rotors of the UAV hit something, it has flipped upside down, etc. 

We should make it so the agent will fail a task if it has excessive forces applied to it, like if the android falls of a cliff or something.

MaxRobinsonTheGreat, 2019-05-21:
We cannot have a robot uprising on our hands

jaydenmilne, 2020-01-16:
We could fail on a certain amount of acceleration too

MaxRobinsonTheGreat, 2020-02-28:
After talking to the oracle @jaydenmilne it seems that it would be better if we added an abuse sensor that simply returns a true or false if the agent has been abused, and the value changes based on which agent it's attached to. The user can then handle how to interpret abuse (negative rewards, terminal, etc.) so we won't mess with the task values.

It might be useful if the sensor returned more complex info, like how hard it was hit or if it flipped over or something like that, but I think a simple abuse sensor is a start.

closed 2020-03-27 by jaydenmilne

jaydenmilne, 2020-03-31:
✅ in changelog