BYU-PCCL/holodeck #263: Enhance Weather Controller

jaydenmilne, 2019-05-21:
**Describe the new feature you'd like**
We should create a weather controller class and change how weather is managed in Holodeck.

There should be a way to specify what the default weather is and if it should be randomized in the config file.

jaydenmilne, 2020-01-09:
We could update the engine and get better skyboxes as well -> https://docs.unrealengine.com/en-US/Support/Builds/ReleaseNotes/4_24/index.html

jaydenmilne, 2020-01-09:
Be sure to update #196 when finished 👆

jaydenmilne, 2020-02-04:
Calling the new class `WeatherController` works but isn't 100% accurate, since it would encompass time methods as well. If we can think of a concise, more descriptive name that'd be great.

vinhowe, 2020-03-11:
Closed by #372, #375, #387.

closed 2020-03-11 by vinhowe

jaydenmilne, 2020-03-31:
✅ in changelog