BYU-PCCL/holodeck #265: Android is Reversed

kolbytn, 2019-05-22:
**Describe the bug**
The android is facing the wrong direction.

**To Reproduce**
First noticed when using the get forward vector in the new Avoid task.

jaydenmilne, 2019-05-24:
Is this just because rotation (#209) is not implemented, so the android is facing the default direction and you can't spin him around?

kolbytn, 2019-05-24:
No, the Android's forward vector comes out of its back. We use actor's forward vectors to calculate line of sight. I think it will take some fiddling with the meshes to fix.

jaydenmilne, 2019-07-26:
[image]

Seems y is forward in stead of X

jaydenmilne, 2019-07-26:
The original mesh from Epic is rotated by 90 degrees. In the blueprints they use that mesh in, it is rotated to face forward again. The problem is that you can't rotate the root component of a blueprint class, and that is the way our AndroidAgent/HandAgent is set up, so we'd need to change the hierarchy of the components.

But, if we do that, there are calls to `GetRootComponent()` all over the place, and they assume that the mesh/actor will be the root component of the blueprint. If we have something else there, it will break (I know this from experience - I accidentally did this the other day).

kolbytn, 2019-07-26:
Is there any way we could override the GetForwardVector method for the Android? Or write a new method to replace GetForwardVector that checks which agent is being used? We'll have to check what the method is actually being called on in the AvoidTask.

MaxRobinsonTheGreat, 2020-02-07:
The avoid task uses the `GetActorForwardVector()`, while other tasks and sensors use `rootcomponent->GetForwardVector()`. It would be most useful to override the latter since the `GetActorForwardVector` relies on it's root component. This could still be problematic since other functions may be affected by the reversed android like `GetVelocity()` that would not be fixed with this hack.

It would be ideal to override `GetGorwardVector()` in the skeletal mesh component, but because this class is a part of the engine, we would have to make another class that inherits from the Skeletal mesh and override the function there. This comes with a whole host of complications, and would basically require that we reconstruct the SkeletalMeshBp. Even if that worked, I tried creating the inherited Skeletal mesh class and overriding the `GetForwardVector()` and could not get it to compile. 

In short, I don't think the override hack would work. It may be necessary to check if a given agent is the android and alter its forward vector accordingly. That would be a mess but I think it may be the best idea so far.