BYU-PCCL/holodeck #331: UAV agent's motors all spin same direction

sethmnielsen, 2019-09-27:
**Describe the bug**
The UAV's propellers are all animated to spin clockwise. 

**To Reproduce**
Please provide sample code to reproduce the bug.

**Expected behavior**
It should be clockwise front-left and back-right, counter-clockwise for front-right and back-left motors.

**Screenshots**
It should look like this:

[image]

I know this doesn't affect anything, I'm just bringing this up for purposes of aesthetics. :)

jaydenmilne, 2019-09-27:
https://www.youtube.com/watch?v=hou0lU8WMgo

MaxRobinsonTheGreat, 2020-01-23:
BYU-PCCL/holodeck-engine#199

closed 2020-01-23 by MaxRobinsonTheGreat

jaydenmilne, 2020-01-23:
white_check_mark in changelog