BYU-PCCL/holodeck #336: TurtleAgent movement is inconsistent between Windows and Linux

jaydenmilne, 2019-11-02:
**Describe the bug**
test_turtleagent_movement is broken on Linux since the turtle agent moves differently. @daniekpo has made a valiant effort to identify the issue.

It appears to be related somehow to the handling of floating point numbers and rounding between Linux and Windows.

 The test has been marked as broken.

**To Reproduce**
Run test_turtleagent_movement. On Windows (the platform it was written on) the test passes; Linux fails.

jaydenmilne, 2020-01-09:
@daniekpo could you describe what you found so we can have it written down before we all forget? 

We've decided this isn't worth fixing.

daniekpo, 2020-01-10:
I collected the agent's location and rotation data from both Windows and Linux. I compared the data when the agent had not moved yet and they seemed pretty equal. I compared the data after moving the agent in a straight line and noticed very minor differences (maybe around e-6) but the differences increased over time. 

I suspect it had something to do with floating-point rounding errors that kept accumulating. It makes sense since we kept reusing the location and rotation to calculate the next location and rotation so the cumulative differences eventually amounted to the agent not reaching its target destination on Linux. That's my theory for now at least.

vinhowe, 2021-04-29:
With #441, we finally have consistency: this test is failing on both Linux and Windows now. We will need to revisit this once it becomes a priority.