BYU-PCCL/holodeck #341: Engine releases loading semaphores without UUID before creating them

jaydenmilne, 2019-11-04:
**Describe the bug**
I'm pretty sure this is bad - [on line 33 of HolodeckServer.cpp](https://github.com/BYU-PCCL/holodeck-engine/blob/master/Source/Holodeck/HolodeckCore/Private/HolodeckServer.cpp#L33):

```cpp
sem_unlink(SEMAPHORE_PATH1);
sem_unlink(SEMAPHORE_PATH2);
auto LoadingSemaphore = sem_open(TCHAR_TO_ANSI(*(LOADING_SEMAPHORE_PATH + UUID)), O_CREAT, 0777, 0);
sem_post(LoadingSemaphore);
LockingSemaphore1 = sem_open(TCHAR_TO_ANSI(*(SEMAPHORE_PATH1 + UUID)), O_CREAT, 0777, 1);
LockingSemaphore2 = sem_open(TCHAR_TO_ANSI(*(SEMAPHORE_PATH2 + UUID)), O_CREAT, 0777, 0);
```

I'm not sure why on Linux we delete semaphore with no UUID and then make it again right after. I feel like this could interfere with other instances of Holodeck

It should either do it with the UUID or not at all.

craig-m-k, 2019-12-03:
FYI I had to comment out these two lines (33 and 34), along with SpawnAgentCommand.cpp:8, to get the ExampleLevel working on openSUSE 15.1, Unreal 4.22.3.

daniekpo, 2019-12-05:
@jaydenmilne I think you're right. It looks like it wasn't changed as the file evolved.
The earliest commit [(bb3851)](https://github.com/BYU-PCCL/holodeck-engine/commit/bb38151c9a01a6e311900a7bed0ddcb05fdb46b5) that I found on that file has these lines (19, 23)
```
// Create the semaphores - Currently destroys existing semaphores
sem_unlink(SEM_PATH1);
sem_unlink(SEM_PATH2);
lockingSemaphore1 = sem_open(SEM_PATH1, O_CREAT, 0777, 1);
lockingSemaphore2 = sem_open(SEM_PATH2, O_CREAT, 0777, 0);
```

[Commit 64e48d0](https://github.com/BYU-PCCL/holodeck-engine/commit/64e48d087e6e0fe38e9cc23a6a9aef81e7af4ddb) changed it to:
```
sem_unlink(SEMAPHORE_PATH1);
sem_unlink(SEMAPHORE_PATH2);
LockingSemaphore1 = sem_open(SEMAPHORE_PATH1, O_CREAT, 0777, 1);
LockingSemaphore2 = sem_open(SEMAPHORE_PATH2, O_CREAT, 0777, 0);
```

[Commit c9c7fa3](https://github.com/BYU-PCCL/holodeck-engine/commit/c9c7fa3f7b352fc3b322b6f3aae2a7f8aeb94445) changed it to
```
sem_unlink(SEMAPHORE_PATH1);
sem_unlink(SEMAPHORE_PATH2);
LockingSemaphore1 = sem_open(*(SEMAPHORE_PATH1 + UUID), O_CREAT, 0777, 1);
LockingSemaphore2 = sem_open(*(SEMAPHORE_PATH2 + UUID), O_CREAT, 0777, 0);
```

It looks like when we switched to appending the UUID we forgot to update the calls to sem_unlink. Although from the documentation, the calls should result in an ENOENT error since the semaphores don't exist. If the UUID is an empty string (which is likely) then this could end up unlinking semaphores that are still in use.

jaydenmilne, 2020-01-07:
We also need to fix the cleanup function, on Windows and Linux it is not using the correct name to unregister them. We should be checking for errors after these system calls and logging them too

daniekpo, 2020-02-26:
Resolved in [holodeck-engine PR 202](https://github.com/BYU-PCCL/holodeck-engine/pull/202)

closed 2020-02-26 by daniekpo