BYU-PCCL/holodeck #342: Loading map in unreal engine failed [holodeck-engine]

insooneelife, 2019-11-06:
**Describe the bug**
A clear and concise description of what the bug is.

I followed all requirements on README.md file, but failed.
Error message after open holodeck.uproject file.

"Failed to load map!
../holodeck-engine/Content/ExampleLevel.umap appears to be an asset file."

**To Reproduce**
Please provide sample code to reproduce the bug.

Just clone the project and follow the manual.

**Expected behavior**
A clear and concise description of what you expected to happen.

I expect the level loaded successfully on unreal engine.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version Information:**
 - OS: [e.g. iOS]
Window 10

 - Version [e.g. 22]
Unreal engine 4.22

 - Holodeck Version
holodeck-engine[master]

 - World/Scenario version

**Additional context**
Add any other context about the problem here.

jaydenmilne, 2019-11-06:
`.uasset` files are stored in git-lfs. From the error you are describing, it seems git-lfs has not converted the pointer file into the actual file. Make sure git-lfs is installed and working on your machine.

Try running `git lfs pull`, or cloning with the https url instead of git.

To verify this I cloned `holodeck-engine` and disabled git-lfs and got the same error:
[image]

jaydenmilne, 2019-11-06:
FYI, most of the assets that we show in our marketing stuff aren't included in `holodeck-engine` for licensing reasons.

closed 2019-11-06 by jaydenmilne