BYU-PCCL/holodeck #343: Unreal engine stops running after click play button. [holodeck-engine]

insooneelife, 2019-11-08:
**Describe the bug**
A clear and concise description of what the bug is.

Unreal engine stops running after click play button.
No response and no clicks.

**To Reproduce**
Please provide sample code to reproduce the bug.

1. Set up holodeck-engine followed by README.md.
2. git lfs pull at holodeck-engine directory. 
3. Open Holodeck.uproject.
4. Play.
Then the editor stops.

**Expected behavior**
A clear and concise description of what you expected to happen.

I expect editor to load level and run just normally.
I guessed some python module could stop the editor, 
so tried to play editor with other map(empty level), 
but it still stops.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version Information:**
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - Holodeck Version
 - World/Scenario version

UE4 4.22

**Additional context**
Add any other context about the problem here.

jaydenmilne, 2019-11-08:
The editor locks up because it is waiting to communicate with a python client.

See [here](https://github.com/BYU-PCCL/holodeck/wiki/Holodeck-Communication-Protocol), particularly the section [Main Synchronization Semaphores](https://github.com/BYU-PCCL/holodeck/wiki/Holodeck-Communication-Protocol#main-synchronization-semaphores)

We have more documentation on how to use the editor and attach a process to it forthcoming, check [Onboarding](https://github.com/BYU-PCCL/holodeck/wiki/Holodeck-Onboarding).

This is assuming that you want to develop the engine. If you just want to use holodeck, you just need to pip install it.

Related to #18 and #223.

closed 2019-11-08 by jaydenmilne

insooneelife, 2019-11-13:
Thank you for the answer.
I saw the link you gave to me, but I could not find the concrete way to attach an instance to running editor. There is no link for it. 
[image]

peterhyun, 2020-01-16:
I'm sorry, but where should I read if I want to develop the engine?

peterhyun, 2020-01-16:
> The editor locks up because it is waiting to communicate with a python client.
>
> See [here](https://github.com/BYU-PCCL/holodeck/wiki/Holodeck-Communication-Protocol), particularly the section [Main [...]

I'm sorry, but where should I read if I want to develop the engine?