BYU-PCCL/holodeck #344: Starting unreal engine 4 editor example causes error. [holodeck-engine]

insooneelife, 2019-11-13:
**Describe the bug**
A clear and concise description of what the bug is.

Starting unreal engine 4 editor example causes error.
https://github.com/BYU-PCCL/holodeck/blob/master/example.py

At this code, if I start with "editor_example()",
I get error like this.
[image]

**To Reproduce**
Please provide sample code to reproduce the bug.
Start holodeck with "editor_example()" in "example.py".

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version Information:**
 - OS: [e.g. iOS]
Window10
 - Version [e.g. 22]
UE4 4.22
 - Holodeck Version
 - World/Scenario version

**Additional context**
Add any other context about the problem here.

jaydenmilne, 2019-11-19:
I can't read the error message - is this a duplicate of #285?

daniekpo, 2019-11-29:
Did you clone the repo? If so, you need to install the package dependencies using pip. Make sure the following packages are installed:
- pywin32
- numpy
- opencv-python

daniekpo, 2019-12-05:
@insooneelife were you able to resolve this issue? Did the suggested fix work?

jaydenmilne, 2020-01-09:
We can't fix it if we can't read the error mesage

closed 2020-01-09 by jaydenmilne