BYU-PCCL/holodeck #346: Can't install on linux

insooneelife, 2019-11-15:
**Describe the bug**
A clear and concise description of what the bug is.

Error install on linux.

[image]

**To Reproduce**
Please provide sample code to reproduce the bug.

Follow up the instructions for linux installation.
https://holodeck.readthedocs.io/en/latest/usage/installation.html

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version Information:**
 - OS: [e.g. iOS]
Linux Ubuntu 16.04

 - Version [e.g. 22]
 - Holodeck Version
 - World/Scenario version

**Additional context**
Add any other context about the problem here.

jaydenmilne, 2019-11-15:
You need to use python3, eg `pip3 install holodeck`

Try installing the dev headers for `python3` too:

`pip3 install python3-dev`

jaydenmilne, 2019-11-19:
Did this resolve your issue?

daniekpo, 2019-11-29:
@insooneelife, were you able to follow the instructions to resolve this issue? Can we close the issue?

jaydenmilne, 2019-12-17:
Assuming this was because the package was not installed.

closed 2019-12-17 by jaydenmilne