BYU-PCCL/holodeck #350: Holodeck Sphere Agent Default Control Scheme

kolbytn, 2019-12-12:
**Describe the bug**
The Holodeck Sphere Agent's default control scheme is the continuous action space, although documentation and file comments claims it should be the discrete one.

**Expected behavior**
Default control scheme (with index 0) should be the discrete action space of size 4.

**Check to make sure all default control schemes are consistent with documentation and comments.

MaxRobinsonTheGreat, 2020-02-07:
How are you running into this bug? You have to add a control scheme field in the scenario config or it won't load the world properly, so there is no "default" control scheme if you're using a config file.

MaxRobinsonTheGreat, 2020-02-10:
Closed for now. I think you were probably using outdated code.

kolbytn, 2020-02-11:
I ran into this issue again and tracked down the discrepancy.

The `SphereAgent` class contains the following property:
```@property
    def control_schemes(self):
        return [("[forward_movement, rotation]", ContinuousActionSpace([2])),
                ("0: Move forward\n1: Move backward\n2: Turn right\n3: Turn left",
                 DiscreteActionSpace([1], 0, 4, buffer_shape=[2]))]
```
However, `ControlSchemes.SPHERE_DISCRETE = 0` and `ControlSchemes.SPHERE_CONTINUOUS = 1`.

MaxRobinsonTheGreat, 2020-03-06:
Fixed with #380

closed 2020-03-06 by MaxRobinsonTheGreat

jaydenmilne, 2020-03-31:
✅ in changelog