BYU-PCCL/holodeck #351: Holodeck Make Scenario Config

kolbytn, 2019-12-12:
**Describe the bug**
When using a python data structure rather than a scenario config file, the following error is thrown. 

```
holodeck/src/holodeck/packagemanager.py", line 345, in get_binary_path_for_scenario
with open(config_path, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'config.json'
```

**To Reproduce**
Call `holodeck.make` with keyword argument `scenario_cfg`

**Expected behavior**
Should load correct world with scenario that was provided as keyword arg

**Additional context**
I believe this is caused by line 62 in holodeck.py

```
scenario_name = "{}-{}".format(scenario["name"], scenario["world"])
```

The scenario name should follow the format `[WORLDNAME]-[SCENARIONAME]`. 

Additionally, I don't think it should be a requirement that the scenario name you are using be an existing scenario name. As is, this is a requirement for holodeck to be able to find the package config file bauase of the logic in `packagemanager.py/get_binary_path_for_scenario`.

daniekpo, 2019-12-17:
I'm currently looking into this issue.

jaydenmilne, 2019-12-17:
> I'm currently looking into this issue.

You can assign yourself, then we know you are working on it (right hand side)

daniekpo, 2019-12-28:
@kolbytn did you say you the documentation specified that you can use any name for the scenario? I think the right behavior is to specify an existing scenario name. I think we may need to fix the documentation.

kolbytn, 2019-12-29:
To my knowledge, the documentation doesn't specify, but I don't think a custom scenario should need to be the same name as an existing scenario file.

daniekpo, 2020-01-06:
I see what you mean. Yeah, that makes sense now that I looked at it again. We need to change how we're getting the executable. Right now it's using the path to an existing scenario to find the executable path.

PR #355 (2020-01-09): Fix/scenario config fixes #351
Fixed the bug where passing a config map object to holodeck.make wasn't working as expected. Updated the documentation. This resolves #351
commit 43ca685e (2020-01-09): Fixed using holodeck.make with a config map. This commit fixes #351
commit b6fee69d (2020-01-09): Merge pull request #355 from BYU-PCCL/fix/scenario_config

Fix/scenario config fixes #351
closed 2020-01-09 by daniekpo