BYU-PCCL/holodeck #354: [holodeck-engine] UAV_ROLL_PITCH_YAW_RATE_ALT in custom worlds

craig-m-k, 2020-01-04:
openSUSE 15.1, Holodeck 0.22, UE 4.22.3

## Background

I am able to use the DefaultWorlds in Linux, but I run into problems with when using custom worlds (e.g. ExampleLevel in holodeck-engine): I always run into the "Semaphore is busy" problem.  I have figured out a way to get around this:

* Within UE4, Play standalone with Additional Launch Parameters:

```
TestWorld -HolodeckOn -LOG=HolodeckLog.txt --HolodeckUUID=a54ce6d6-db11-4a7f-acb9-cd37f822a205
```

* Comment out lines 33 and 34 of HolodeckServer.cpp.

After this, I call a custom holodeck.make function, where param_dict['uuid'] is set to the same uuid as above, and start_world is False. Note: /dev/shm must be cleaned of this uuid for this to work.

## The problem

This setup works for me with UAV_TORQUES control scheme.  But if I change either control_scheme to 1 in the JSON file, or manually set it via env.agent, I get the following error in UE4:

```
LogHolodeck: Error: UUAVControlSchemeTargetRollPitch couldn't get Uav reference
```

The relevant snippet from UavControlSchemeTargetRollPitch.cpp:33

```
if (Uav == nullptr) {
    Uav = static_cast<AUav*>(UavController->GetPawn());
    if (Uav == nullptr) {
        UE_LOG(LogHolodeck, Error, TEXT("UUAVControlSchemeTargetRollPitch couldn't get Uav reference"));
        return;
    }
}
```

This does not crash the program, but step and act seem to not send any commands to the UAV.

When I run other environments (e.g. EuropeanForest-MaxDistance or ExampleLevel) in the "usual" way (i.e. invoked via holodeck.make as in examples.py), I don't have any problems with either ControlScheme.  This is strange because I would think there wouldn't be any difference between ExampleLevel invoked in UE4 as a standalone game versus running ExampleLevel provided in the Holodeck package from the command line.  Any thoughts?

jaydenmilne, 2020-01-07:
I'll look into this, but I'm just curious as to how you are using a custom build with the worlds we provide in the built package. It doesn't immediately seem related, but we did just fix an issue with control schemes on Linux, maybe try applying this commit and seeing if it helps: BYU-PCCL/holodeck-engine#198

BTW lines 33 and 34 are wrong, those are just some #ifdef bit rot, we're fixing it.

jaydenmilne, 2020-03-31:
Any updates on this issue? I haven't been able to find anything

vinhowe, 2021-04-29:
Closing for now as https://github.com/BYU-PCCL/holodeck-engine/commit/aa8ddbece013a55ca5e8b25c88fe8a2919d35c0e seems to have fixed this.

closed 2021-04-29 by vinhowe