BYU-PCCL/holodeck #356: Leaking File Descriptors (`OSError: [Errno 24] Too many open files`)

jaydenmilne, 2020-01-09:
**Describe the bug**
When `holodeck.make()` is run repeatedly, it leaks file descriptors. Eventually it will crash with `OSError: [Errno 24] Too many open files`

**To Reproduce**
```python
import holodeck
from tqdm import tqdm

for _ in tqdm(range(1000)):
    with holodeck.make("MazeWorld-FinishMazeSphere") as env:
        pass
```

This will crash at around 13 iterations

**Expected behavior**
A clear and concise description of what you expected to happen.

**Stack Trace**

```
jayden@naga:~/holodeck$ python3 repo-bug.py 
Traceback (most recent call last):
  File "repo-bug.py", line 4, in <module>
  File "/home/jayden/holodeck/src/holodeck/holodeck.py", line 87, in make
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 134, in __init__
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 285, in reset
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 250, in _load_scenario
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 408, in add_agent
  File "/home/jayden/holodeck/src/holodeck/agents.py", line 170, in add_sensors
  File "/home/jayden/holodeck/src/holodeck/sensors.py", line 650, in build_sensor
  File "/home/jayden/holodeck/src/holodeck/sensors.py", line 34, in __init__
  File "/home/jayden/holodeck/src/holodeck/holodeckclient.py", line 120, in malloc
  File "/home/jayden/holodeck/src/holodeck/shmem.py", line 42, in __init__
OSError: [Errno 24] Too many open files: '/dev/shm/HOLODECK_MEM5ff7a9cb-26c4-4f1a-a473-4690257e75d4_sphere0_DistanceTask_sensor_data'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 780, in get_code
  File "<frozen importlib._bootstrap_external>", line 832, in get_data
OSError: [Errno 24] Too many open files: '/usr/lib/python3/dist-packages/apport/__init__.py'

Original exception was:
Traceback (most recent call last):
  File "repo-bug.py", line 4, in <module>
  File "/home/jayden/holodeck/src/holodeck/holodeck.py", line 87, in make
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 134, in __init__
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 285, in reset
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 250, in _load_scenario
  File "/home/jayden/holodeck/src/holodeck/environments.py", line 408, in add_agent
  File "/home/jayden/holodeck/src/holodeck/agents.py", line 170, in add_sensors
  File "/home/jayden/holodeck/src/holodeck/sensors.py", line 650, in build_sensor
  File "/home/jayden/holodeck/src/holodeck/sensors.py", line 34, in __init__
  File "/home/jayden/holodeck/src/holodeck/holodeckclient.py", line 120, in malloc
  File "/home/jayden/holodeck/src/holodeck/shmem.py", line 42, in __init__
OSError: [Errno 24] Too many open files: '/dev/shm/HOLODECK_MEM5ff7a9cb-26c4-4f1a-a473-4690257e75d4_sphere0_DistanceTask_sensor_data'
```

**Version Information:**
 - OS: Ubuntu
 - Version 0.3.0

**Additional Context**
It should be cleaning up all of the descriptors with the `HolodeckEnvirontment.__exit__` -> `HolodeckClient.unlink` -> `Shmem.unlink`

jaydenmilne, 2020-03-24:
See https://github.com/BYU-PCCL/holodeck/blob/5d86a6c713519a57ea21bace6c58ed70cc62ede3/src/holodeck/shmem.py#L47

daniekpo, 2020-05-06:
@jaydenmilne, @vinhowe, can you still repro this issue? I could not. It ran the entire script without throwing the exception.

vinhowe, 2020-05-06:
Have you tried running the tests on the Docker image? That's where I was
seeing it most consistently, although I've seen it on the host too.

daniekpo, 2020-05-06:
> Have you tried running the tests on the Docker image? That's where I was seeing it most consistently, although I've seen it on the host too.
> […](#)

I was able to repro by reducing the open file limit on the host.

daniekpo, 2020-05-06:
I was able to repro the issue (although I didn't get the exact exception message) by reducing the open files limit on the host.

daniekpo, 2020-05-11:
Resolved in #418

PR #418 (2020-05-11): Fix the max open files error
closes #356
closes #407

It turns out that calling os.remove or unlinking a mapped memory file might delete the actual file but the OS keeps the mapped file open until there's no reference to that memory block. We were removing the mapped memory files we created, but since we still kept references to the location in memory, the OS left the file (handle) open. Running multiple Holodeck instances eventually in one process eventually exhausts the open file limit.

----
To verify that this fixed the problem, you can run the script below:
First set the open files soft l [... cut]
commit 07d232ea (2020-05-11): Fix the max open files error

It turns out that calling os.remove or unlinking a
mapped memory file might delete the actual file
but the OS keeps the mapped file open until there's
no reference to that memory block. We were removing
the mapped memory files we created, but since we
still kept references to the location in memory,
the OS did not close the file (handle). Running multiple
Holodeck instances eventually in one process eventually
exaust the open file limit.

This closes #356 and also closes #407
closed 2020-05-11 by daniekpo