BYU-PCCL/holodeck #357: Add new agents myself?

peterhyun, 2020-01-16:
Reading the documents, I don't think it's a feature yet to add new agents by myself other than the 6 default agents given in the UE4 project file. It would be nice if it's possible.

jaydenmilne, 2020-01-16:
We are looking to make it easier and document the process in the next month or so, currently there is a lot of boilerplate you have to write. When we do finish it you'll find it on the [developer wiki](https://github.com/BYU-PCCL/holodeck/wiki/Holodeck-Onboarding), not the [docs](https://holodeck.readthedocs.io/en/latest/)

We consider this a more advanced use case however since you do have to re-build and package the engine if you want to add an agent, which we aren't expecting most users to do.

We'll keep this open to remind ourselves to write this documentation, thanks.

vinhowe, 2020-03-18:
@jaydenmilne Are you working on this?