BYU-PCCL/holodeck #358: ExampleLevel bug? How am I supposed to use this file?

peterhyun, 2020-01-16:
I'm trying to run the example.py file's editor_example() function. I realized running this function without playing ExampleLevel gives me an error

>   File "c:\git\clone\holodeck\src\holodeck\holodeckclient.py", line 47, in __windows_init__
>     "Global\\HOLODECK_SEMAPHORE_SERVER" + self._uuid)
> pywintypes.error: (2, 'OpenSemaphore', 'The system cannot find the file specified.')

which is expected.

Before playing the ExampleLevel, I dragged the UAVBlueprint file(In Content/HolodeckContent/Agents/UAV) within the ExampleLevel world. Not sure if this is the correct way, but I thought I had to cause editor_example uses an UavAgent type agent named "uav0". Then I pressed the play button, but then UE4 4.22 crashes and gives me the report

> Assertion failed: [File:C:\git\clone\holodeck-engine\Source\Holodeck\HolodeckCore\Private\HolodeckPawnController.cpp] [Line: 38] HolodeckPawnController couldn't find server...

How can I run the example.py's editor_example() function successfully?

jaydenmilne, 2020-01-16:
Thanks for pointing those out, those last two examples are out of date. Previously there would be agents baked into the world (dragged in as you did). However, that limited each world to one particular agent, so made it so agents were spawned dynamically based on a configuration given to the client, which allowed more flexibility.

Here is a fixed version, I'll have this merged soonish
```python
def editor_example():
    """This editor example shows how to interact with holodeck worlds while they are being built
    in the Unreal Engine Editor. Most people that use holodeck will not need this.

    This example uses a custom scenario, see 
    Note: When launching Holodeck from the editor, press the down arrow next to "Play" and select
    "Standalone Game", otherwise the editor will lock up when the client stops ticking it.
    """

    config = {
        "name": "test",
        "world": "TestWorld",
        "main_agent": "uav0",
        "agents": [
            {
                "agent_name": "uav0",
                "agent_type": "UavAgent",
                "sensors": [
                    {
                        "sensor_type": "LocationSensor",
                    },
                    {
                        "sensor_type": "VelocitySensor"
                    },
                    {
                        "sensor_type": "RGBCamera"
                    }
                ],
                "control_scheme": 1,
                "location": [0, 0, 1]
            }
        ]
    }

    env = HolodeckEnvironment(scenario=config, start_world=False)
    command = [0, 0, 10, 50]

    for i in range(10):
        env.reset()
        for _ in range(1000):
            state, reward, terminal, _ = env.step(command)
```

peterhyun, 2020-01-17:
Thanks for the reply! May I ask you one more question?
I'm trying to add new agents and tasks as I mentioned in my previous post, and I assume I have to modify agents.py, task.py, and add couple of files in the c++ files within the Agents folder and Tasks folder. However, I can't find out when the DistanceTask.cpp file is called when running editor_example() from example.py. Who's calling the functions in DistanceTask.cpp?

jaydenmilne, 2020-01-21:
There is no DistanceTask in `editor_example()`. A task is simply another sensor on the agent, so if you want the DistanceTask you need to add one to the `sensors` array. See [here](https://github.com/BYU-PCCL/holodeck-configs/blob/0e8a8a08d00a181caa515281ac748acead96bc8c/DefaultWorlds/AndroidPlayground-MaxDistance.json#L36) for an example.

jaydenmilne, 2020-01-21:
Here is the documentation for tasks:
https://holodeck.readthedocs.io/en/latest/packages/docs/tasks/tasks.html

jaydenmilne, 2020-01-23:
Docs are fixed.

closed 2020-01-23 by jaydenmilne