BYU-PCCL/holodeck #361: Editor crashes when an agent it manually added

MaxRobinsonTheGreat, 2020-01-23:
**Describe the bug**
The engine crashes when an agent is manually placed in a world rather than dynamically spawned. Though this is no longer a common use case, it should be fixed to avoid future confusion and potential use cases.

**To Reproduce**
Drag and drop a uav into the example level. Start the game and it will crash.

**Expected behavior**
The engine should not crash.

**Version Information:**
 - OS: Linux and Windows
 - Holodeck version: 0.3.0

**Additional context**
This is likely caused by the agent not being properly initialized. We should make sure all the initialization functions are being called.

MaxRobinsonTheGreat, 2020-02-10:
#366

closed 2020-02-10 by MaxRobinsonTheGreat

jaydenmilne, 2020-03-31:
✅ in changelog