BYU-PCCL/holodeck #369: Document Torque Limits for Agents

jaydenmilne, 2020-01-30:
**Describe the new feature you'd like**
For every control scheme 

- [ ] Document Android control schemes (see [HandAgent](https://holodeck.readthedocs.io/en/latest/agents/hand-agent.html#control-schemes))
- [ ] Document UAV control scheme maximum values
- [ ] Look into adding scaled thrust control scheme for UAV
- [ ] Look at other agents

If you have questions ask @kolbytn

kolbytn, 2020-02-12:
It would be great if these limits were available in the `ContinuousActionSpace`. Additionally, the `_low` and `_high` properties of the `DiscreteActionSpace` class should be made accessible.

daniekpo, 2020-03-04:
Resolved. See PR #380

closed 2020-03-04 by daniekpo

jaydenmilne, 2020-03-31:
✅ in changelog

PR #380 (2020-03-04): Android control scheme documentation. Control scheme min/max
Added methods in agent to get the min and max values of the control scheme. Added tests. Fixes #369