BYU-PCCL/holodeck #373: Android Joint Movement Inconsistent Between Linux and Windows

jaydenmilne, 2020-02-06:
**Describe the bug**
Running [test_joint_rotation_sensor]( fails much more on Linux than it does on Windows. From comparing the movement of the Android in the viewport, it seems that the Android moves a lot less on Linux vs Windows. 

This means that most joints are unable to fully exercise their range of movement, and the test fails.

If I was a betting man I would say this is related to #336.

**To Reproduce**
Run the test on Linux and on Windows. 

**Expected behavior**
The Android should behave the same.

**Version Information:**
 - Holodeck Version: Reproduces on `0.3.0` and develop

jaydenmilne, 2020-02-06:
[Joints failing on Linux](https://pastebin.com/d6L5duZs)

daniekpo, 2020-02-07:
I thought this was fixed. I haven't seen it fail since [this fix.](https://github.com/BYU-PCCL/holodeck-engine/pull/198).

jaydenmilne, 2020-02-11:
Duplicate of #198. We should add that to the changelog

jaydenmilne, 2020-02-11:
Duplicate of BYU-PCCL/holodeck-engine#198, I was using an older build of develop. We should add that to the changelog if it is present in 0.3.0 @daniekpo

closed 2020-02-11 by jaydenmilne