BYU-PCCL/holodeck #402: Abount holodeck-engine RelativeSkeletalPositionSensor

insooneelife, 2020-03-20:
Hello guys
I found a weird code in RelativeSkeletalPositionSensor.cpp

void URelativeSkeletalPositionSensor::TickSensorComponent(float DeltaTime, ELevelTick TickType, FActorComponentTickFunction* ThisTickFunction) {
	float* FloatBuffer = static_cast<float*>(Buffer);
	for (int i = 0; i < Bones.Num(); i++) {

		//UE_LOG(LogTemp, Warning, TEXT("bone idx : %d  name : %s"), i, *Bones[i].ToString());

		FQuat BoneQ = SkeletalMeshComponent->GetBoneQuaternion(Bones[i], EBoneSpaces::WorldSpace);
		FQuat ParentQ = SkeletalMeshComponent->GetBoneQuaternion(ParentBones[i], EBoneSpaces::WorldSpace);
		FQuat Quat = ParentQ.Inverse() * BoneQ;
		FloatBuffer[4 * i] = Quat.X;
		FloatBuffer[4 * i + 1] = Quat.Y;
		FloatBuffer[4 * i + 2] = Quat.Z;
		FloatBuffer[4 * i + 3] = Quat.W;
	}
}

int URelativeSkeletalPositionSensor::GetNumItems() {
	return this->Bones.Num();
}

at the code, TickSensorComponent()
~
FloatBuffer[4 * i] = Quat.X;
FloatBuffer[4 * i + 1] = Quat.Y;
FloatBuffer[4 * i + 2] = Quat.Z;
FloatBuffer[4 * i + 3] = Quat.W;
~
they are accessing array index up to  BoneNum*4,

but in the code, GetNumItems()
it just returns Bone.Num().

Shouldn't this code be changed like this?
int URelativeSkeletalPositionSensor::GetNumItems() {
	return this->Bones.Num() * 4;
}

jaydenmilne, 2020-03-31:
Yes, that probably should be changed, but it currently isn't causing any issues. That `GetNumItems` is used to allocate the size of the shared memory buffer if the client hasn't already done so. Since the client normally runs first, [and it's calculation is correct](https://github.com/BYU-PCCL/holodeck/blob/1eb444e0a142c9203cd56786d0946dd773adbea1/src/holodeck/sensors.py#L456), we haven't seen anything.

Thank you for pointing this out.

We should spot check the rest of the sensors to make sure that they are the same on both sides.

daniekpo, 2020-05-06:
@jaydenmilne, from your comment above, is this a bug?