BYU-PCCL/holodeck #406: posix_ipc.SignalError When making environment

jmangelson, 2020-03-27:
When making an environment a posix_ipc.SignalError occures. 

Once running the env = holodeck.make("UrbanCity-MaxDistance") command, the error below is immediately thrown. 

The Viewport loads, but because the creation of the environment fails the env object does not exist. 

>>> import holodeck
>>> import numpy as np
>>> env = holodeck.make("UrbanCity-MaxDistance")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/joshua/.local/lib/python3.5/site-packages/holodeck/holodeck.py", line 87, in make
    return HolodeckEnvironment(**param_dict)
  File "/home/joshua/.local/lib/python3.5/site-packages/holodeck/environments.py", line 99, in __init__
    show_viewport=show_viewport)
  File "/home/joshua/.local/lib/python3.5/site-packages/holodeck/environments.py", line 626, in __linux_start_process__
    loading_semaphore.acquire(10)
posix_ipc.SignalError: The wait was interrupted by a signal

jmangelson, 2020-03-27:
This discussion seems relevent:

https://groups.google.com/forum/#!topic/comp.lang.python/raOemE38PaY

jaydenmilne, 2020-03-27:
What version of Linux are you using?

jmangelson, 2020-03-27:
I'm using Ubuntu 16.04

Joshua Mangelson, Ph.D.
Post-Doctoral Research Fellow
Robotic Perception Lab
Carnegie Mellon University, RI
http://robots.engin.umich.edu/~joshuagm/

jaydenmilne, 2020-03-27:
Hmm, I wonder if this is an exception that we just need to swallow and ignore. Presumably the Python interpreter receiving a signal shouldn’t impact waiting on a semaphore for us, we just need to start the wait over again. I wonder what signal is being received, we’ve never seen this before, and we developed Holodeck on Ubuntu 16.04. I wonder what is different about your system.  From: jmangelsonSent: Friday, March 27, 2020 14:40To: BYU-PCCL/holodeckCc: Jayden Milne; CommentSubject: Re: [BYU-PCCL/holodeck] posix_ipc.SignalError When making environment (#406) I'm using Ubuntu 16.04Joshua Mangelson, Ph.D.Post-Doctoral Research FellowRobotic Perception LabCarnegie Mellon University, RIhttp://robots.engin.umich.edu/~joshuagm/

jmangelson, 2020-03-27:
> robots.engin.umich.edu/~joshuagm/On Fri, Mar 27, 2020, 4:38 PM Jayden

I think you are right that we potentially can just swallow the exception.

Joshua Mangelson, Ph.D.
Post-Doctoral Research Fellow
Robotic Perception Lab
Carnegie Mellon University, RI
http://robots.engin.umich.edu/~joshuagm/

jmangelson, 2020-03-27:
I'm not sure what is different on my machine. Its been a while since I did
the ubuntu install, so it could be multiple things.

Joshua Mangelson, Ph.D.
Post-Doctoral Research Fellow
Robotic Perception Lab
Carnegie Mellon University, RI
http://robots.engin.umich.edu/~joshuagm/

jmangelson, 2020-03-27:
So, I tried modifying line 630 in environments.py to ignore the exception:
        except posix_ipc.SignalError:
            pass

It now complains with a semaphore error:

>>> import holodeck
>>> env = holodeck.make("UrbanCity-MaxDistance")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File
"/home/joshua/.local/lib/python3.5/site-packages/holodeck/holodeck.py",
line 87, in make
    return HolodeckEnvironment(**param_dict)
  File
"/home/joshua/.local/lib/python3.5/site-packages/holodeck/environments.py",
line 106, in __init__
    self._client = HolodeckClient(self._uuid, start_world)
  File
"/home/joshua/.local/lib/python3.5/site-packages/holodeck/holodeckclient.py",
line 35, in __init__
    self.__posix_init__()
  File
"/home/joshua/.local/lib/python3.5/site-packages/holodeck/holodeckclient.py",
line 70, in __posix_init__
    self._semaphore1 = posix_ipc.Semaphore("/HOLODECK_SEMAPHORE_SERVER" +
self._uuid)
posix_ipc.ExistentialError: No semaphore exists with the specified name

Joshua Mangelson, PhD
Post-Doctoral Research Fellow
Robotic Perception Lab
Field Robotics Center, RI
Carnegie Mellon University
http://robots.engin.umich.edu/~joshuagm/

jmangelson, 2020-03-28:
I tried it on a clean install of 18.04 and it works fine.

jaydenmilne, 2020-03-31:
Strange. Well I'm glad you got it working, let's hope it doesn't come up again

closed 2020-03-31 by jaydenmilne