BYU-PCCL/holodeck #413: CyberPunkCity-Follow Agent ActionSpace

kolbytn, 2020-04-08:
**Describe the bug**
Action spaces for both agents on this config return a list of `None` for `get_high` and `get_low`

**To Reproduce**
Run `CyberPunkCity-Follow` scenario

**Expected behavior**
`get_high` and `get_low` should return a list of float values associated with those agents

**Version Information:**
 - Ubuntu
 - Holodeck Version 0.3.1

daniekpo, 2020-04-13:
@kolbytn `UavAgent` doesn't have any restrictions on the second control scheme that's why we return None. I think -inf and inf would probably be better. I could not repro the bug on the second agent in that scenario (`nav0`). I got the correct high and low values.

kolbytn, 2020-04-13:
Oh I see it now. Not sure why I was getting that for the nav agent. You can close this. `inf` may be better, but I'll leave that up to you.

daniekpo, 2020-05-11:
Fixed in #417

PR #417 (2020-05-07): Fix uav min max
- Changed UAV agent's target action space max and min to infinity and negative infinity respectively
- Should make it more obvious that there are no bounds instead of using `None`
- Closes #413
closed 2020-05-11 by daniekpo