BYU-PCCL/holodeck #414: Holodeck continues running after process closes

kolbytn, 2020-04-08:
**Describe the bug**
The holodeck backend continues running after the process terminates. Not sure in what cases this happens, but it probably has something to do with threading.

[image]

**To Reproduce**
Run holodeck using the repo found at `https://github.com/kolbytn/holodeck-baselines`. Then quit the process. Holodeck should still be running.

**Expected behavior**
The holodeck backend should exit when the frontend exits.

**Version Information:**
 - Ubuntu
 - Holodeck Version 0.3.1

daniekpo, 2020-04-13:
@kolbytn I haven't been able to repro this bug

jaydenmilne, 2020-04-13:
Have you been using the repo he linked?

daniekpo, 2020-04-13:
Yeah! I ran it a few times and checked `top` but couldn't find any holodeck process

kolbytn, 2020-04-13:
That's odd because it was happening consistently to me. And you're using Ubuntu? Make sure to run `main.py` long enough for all the environments to load up. You can use `--viewport` so that the environments will display.

daniekpo, 2020-04-14:
Oh! Maybe that's why. I was running it from a remote session. I'll have to set up a Linux environment on my local machine coz my network isn't fast enough for any display forwarding/remote desktop method that I've tried

kolbytn, 2020-04-14:
I saw the issue ssh'd into a lab machine as well. You shouldn't need to run locally. Just make sure you let it run for long enough to see Holodeck appear on `htop` in the first place.

daniekpo, 2020-05-05:
Ok. I think I accidentally left it running for about an hour and killed main.py, but I still could not see traces of Holodeck. How long did you have it running?

kolbytn, 2020-05-06:
That should have been enough time. I was only running for maybe 2-5 minutes each time, but I ran it repeatedly... You could try running it for a few minutes, killing the process, and then spinning up a new one. Do that several times and see if you can see anything. I'll try again myself when I can.

daniekpo, 2020-05-11:
@kolbytn, I was finally able to reproduce it. I was sending it SIGINT by doing ctrl-c, but I think what you were talking about was killing/exiting the terminal which sent SIGHUB. Either way, I installed signal handlers for those events.

kolbytn, 2020-05-12:
Awesome thanks! I thought I was doing ctrl-c over ssh, but I may be mistaken. Hopefully that problem is gone for good.

daniekpo, 2020-05-15:
Resolved in #419

PR #419 (2020-05-14): Install a handler for SIGHUB
We were not handling SIGHUB previously so resources were not freed up when the terminal was killed/closed. This commit fixes those issues.

Closes #414
commit ee04a6d8 (2020-05-14): Install a handler for SIGHUB

We were not handling SIGHUB previously so resources were
not freed up when the terminal was killed/closed.
Closes #414
closed 2020-05-15 by daniekpo