BYU-PCCL/holodeck #422: NavAgent doesn't seem to be working

sethmnielsen, 2020-05-23:
**Describe the bug**
I can get the NavAgent to spawn in the world, but it doesn't seem to respond to any commands given to it. I tried both the `CyberPunkCity` and `UrbanCity` worlds.

**To Reproduce**
I first used the default `CyberPunkCity-Follow` scenario, then I used the same scenario with two changes:

- for `uav0`: changed `control_scheme` to `1`
- for `nav0`: changed `rotation` to `[0.0, 0.0, 0.0]`

 and copied the code from the [Multi Agent Example](https://holodeck.readthedocs.io/en/latest/usage/examples/multi-agent.html) in the documentation:
```python
import holodeck
import numpy as np

env = holodeck.make('CyberPunkCity-Follow')
env.reset()

env.act('uav0', np.array([0, 0, 0, 100]))
env.act('nav0', np.array([0, 0, 0]))
for i in range(300):
   states = env.tick()

   # states is a dictionary
   task = states["uav0"]["FollowTask"]

   reward = task[0]
   terminal = task[1]
   location = states["uav0"]["LocationSensor"]
```
I then pressed `tab` to go to the NavAgent's location and then `v` to look around.

**Expected behavior**
I would expect the NavAgent to try to move to the location `[0, 0, 0]`, but it doesn't do anything. The default config has it spawn rotated where it is mostly stuck below the ground, so I changed its initial rotation to all zeros in the json file. Still, the NavAgent just stood there without doing anything. Also, it seems like gravity doesn't affect the NavAgent either, because it will float in the air wherever it spawns without ever falling.

Any ideas on how to fix this?

**Logs**
[HolodeckLog.txt](https://github.com/BYU-PCCL/holodeck/files/4672701/HolodeckLog.txt)

**Version Information:**
 - Ubuntu 18.04.4
 - Holodeck Version: 0.3.1

MaxRobinsonTheGreat, 2020-05-28:
Thanks for documenting this. The navagent has received very little support recently and is in need of some upkeep. We will add this to our todo! (Also, the nav agent is not meant to be affected by gravity, it is constricted to its xy plane)

Screenshot of nav agent in CyberPunkCity-Follow:
[image]

sethmnielsen, 2020-06-02:
Cool, glad to see that get added to the 0.3.2 milestone. Do you know if there is any quick-ish fix that I can do on my end in the meantime? Dr. Beard and I were hoping to use the NavAgent in a target-following assignment next week for his class. It isn't the end of the world if we can't - we could probably do something with a second UAV as the target instead. That being said, following a person-like target would be ideal!

sethmnielsen, 2020-10-15:
Any update on this issue? Man, that NavAgent would be super useful...