BYU-PCCL/holodeck #424: Signal Handling breaks on Windows

MaxRobinsonTheGreat, 2020-05-28:
**Describe the bug**
When running holodeck on windows the environment throws the below error. Windows does not have signal.SIGHUP. The problem is temporarily fixed by commenting out the line `signal.signal(signal.SIGHUP, self.graceful_exit)` but this seems to defeat the purpose of the signal handler.

**To Reproduce**
Create an environment in Windows

**Logs**
```
Traceback (most recent call last):
  File ".\example.py", line 33, in <module>
    uav_example()
  File ".\example.py", line 12, in uav_example
    env = holodeck.make("CleanUp-TableHand")
  File "c:\users\max\desktop\holodeck\src\holodeck\holodeck.py", line 91, in make
    return HolodeckEnvironment(**param_dict)
  File "c:\users\max\desktop\holodeck\src\holodeck\environments.py", line 143, in __init__
    signal.signal(signal.SIGHUP, self.graceful_exit)
AttributeError: module 'signal' has no attribute 'SIGHUP'
```
**Version Information:**
 - Windows 10
 - Holodeck Version 0.3.2dev

daniekpo, 2020-05-28:
Oh, shoot! I totally forgot about Windows when I implemented that. I'll fix it

daniekpo, 2020-05-29:
Fixed in #426

PR #426 (2020-05-29): Fix event handling error on Windows
Closes #424
closed 2020-05-29 by daniekpo