BYU-PCCL/holodeck #428: environment issue

sungreong, 2020-08-14:
**Describe the bug**
It is not working. please help me

**To Reproduce**
Please provide sample code to reproduce the bug.
```python
command = np.array([0, 0, 0, 100])
env.reset()
for _ in range(180):
   state, reward, terminal, info = env.step(command)
```
**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
```
---------------------------------------------------------------------------
BusyError                                 Traceback (most recent call last)
~/anaconda3/envs/pytorch/lib/python3.7/site-packages/holodeck/environments.py in __linux_start_process__(self, binary_path, task_key, gl_version, verbose, show_viewport)
    626         try:
--> 627             loading_semaphore.acquire(10)
    628         except posix_ipc.BusyError:

BusyError: Semaphore is busy

During handling of the above exception, another exception occurred:

HolodeckException                         Traceback (most recent call last)
<ipython-input-5-7df39fdb4b7a> in <module>
      2 import numpy as np
      3 
----> 4 env = holodeck.make("UrbanCity-MaxDistance")
      5 
      6 # The UAV takes 3 torques and a thrust as a command.

~/anaconda3/envs/pytorch/lib/python3.7/site-packages/holodeck/holodeck.py in make(scenario_name, scenario_cfg, gl_version, window_res, verbose, show_viewport, ticks_per_sec, copy_state)
     89         param_dict["window_size"] = window_res
     90 
---> 91     return HolodeckEnvironment(**param_dict)

~/anaconda3/envs/pytorch/lib/python3.7/site-packages/holodeck/environments.py in __init__(self, agent_definitions, binary_path, window_size, start_world, uuid, gl_version, verbose, pre_start_steps, show_viewport, ticks_per_sec, copy_state, scenario)
    100             if os.name == "posix":
    101                 self.__linux_start_process__(binary_path, world_key, gl_version, verbose=verbose,
--> 102                                              show_viewport=show_viewport)
    103             elif os.name == "nt":
    104                 self.__windows_start_process__(binary_path, world_key, verbose=verbose)

~/anaconda3/envs/pytorch/lib/python3.7/site-packages/holodeck/environments.py in __linux_start_process__(self, binary_path, task_key, gl_version, verbose, show_viewport)
    627             loading_semaphore.acquire(10)
    628         except posix_ipc.BusyError:
--> 629             raise HolodeckException("Timed out waiting for binary to load. Ensure that holodeck is "
    630                                     "not being run with root priveleges.")
    631         loading_semaphore.unlink()

HolodeckException: Timed out waiting for binary to load. Ensure that holodeck is not being run with root priveleges.

```
**Version Information:**
 - Platform / Version [e.g. Ubuntu 16.04]
 - Holodeck Version (output of `holodeck.util.get_holodeck_version(0.3.1')`)

hwgao1101, 2023-10-24:
Have you ever solved this problem? I meet the same question.
Traceback (most recent call last):
  File "/home/titanium/anaconda3/envs/neusis/lib/python3.7/site-packages/holoocean/environments.py", line 783, in __linux_start_process__
    loading_semaphore.acquire(10)
posix_ipc.BusyError: Semaphore is busy
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/opt/pycharm/pycharm-community-2023.1.3/plugins/python-ce/helpers/pydev/pydevd.py", line 1496, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/opt/pycharm/pycharm-community-2023.1.3/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/titanium/1ubuntu/Holosimulation/uuv-sonnar-simulator/myTankSimulator.py", line 39, in <module>
    with holoocean.make(scenario) as env:
  File "/home/titanium/anaconda3/envs/neusis/lib/python3.7/site-packages/holoocean/holoocean.py", line 103, in make
    return HoloOceanEnvironment(**param_dict)
  File "/home/titanium/anaconda3/envs/neusis/lib/python3.7/site-packages/holoocean/environments.py", line 158, in __init__
    show_viewport=show_viewport)
  File "/home/titanium/anaconda3/envs/neusis/lib/python3.7/site-packages/holoocean/environments.py", line 785, in __linux_start_process__
    raise HoloOceanException("Timed out waiting for binary to load. Ensure that holoocean "
holoocean.exceptions.HoloOceanException: Timed out waiting for binary to load. Ensure that holoocean is not being run with root priveleges.
python-BaseException

taeseongkr, 2024-11-21:
> hwgao1101

I've seen your work with HoloOcean on your Git. Could you share what Ubuntu version you used?