BYU-PCCL/holodeck #438: Misalignment in Aspects of Uav Agent

chris-luangrath, 2020-12-19:
**Describe the bug**
The MaxHeightCeiling() function is called in HolodeckAgent.cpp. The way it works is if the Z coordinate of an agent is above the set max height, it will teleport the agent down to the max height. However, the sensors return that the height is _slightly_ above the max height. For example, if the set max height is 1, the sensor records that at the agent is at 1.0692639e+00 when it's at the top. I think this is only testable with the Uav agent because I don't know if there are commands to propel other agents upwards. I'll have to look into that later. It could also have something to do with gravity, but I don't know which agents are affected by gravity and which ones aren't.

**To Reproduce**
To reproduce this, you will need a holodeck-engine that has the MaxHeightCeiling() function working, so you probably will need the max-height build of the holodeck-engine if you want to test this (unless it's already merged into develop).
```python
import holodeck
from holodeck import agents
from holodeck.environments import *
from holodeck import sensors

def uav_max_height():
    """This editor example shows how to interact with holodeck worlds while they are being built
    in the Unreal Engine Editor. Most people that use holodeck will not need this.

    This example uses a custom scenario, see
    https://holodeck.readthedocs.io/en/latest/usage/examples/custom-scenarios.html

    Note: When launching Holodeck from the editor, press the down arrow next to "Play" and select
    "Standalone Game", otherwise the editor will lock up when the client stops ticking it.
    """

    config = {
        "name": "test",
        "world": "TestWorld",
        "main_agent": "agent0",
        "agents": [
            {
                "agent_name": "agent0",
                #"agent_type": "AndroidAgent",
                #"agent_type": "HandAgent",
                #"agent_type": "NavAgent",
                #"agent_type": "SphereAgent",
                #"agent_type": "TurtleAgent",
                "agent_type": "UavAgent",
                "sensors": [
                    {
                        "sensor_type": "LocationSensor",
                    },
                    {"sensor_type": "VelocitySensor"},
                    {"sensor_type": "RGBCamera"},
                ],
                "control_scheme": 0,
                "location": [0, 0, 5],
                "max_height": 1,

            }
        ],
    }

    env = HolodeckEnvironment(scenario=config, start_world=False)

    command = [0, 0, 0, 1000]
    env.act("agent0", command)
    state = env.tick(50)

    print(state["LocationSensor"])

uav_max_height()
```
**Expected behavior**
I'm pretty sure that if the max height is set, the Uav shouldn't ever go over that height.

**Logs**
```
PS C:\Users\Chris\Documents\holodeck> python example.py
[-4.1723251e-08 -1.6298145e-10  1.0692639e+00]
```
**Version Information:**
 - Version	10.0.19042 Build 19042
 - Holodeck Version (output of `holodeck.util.get_holodeck_version()`) (I don't know how to do this)

vinhowe, 2021-01-10:
Could you put this change into a PR off of `max-height`? I might not have enough background (maybe @MaxRobinsonTheGreat does) but I don't know if I can tell from your output whether the thrust change is helping.

vinhowe, 2021-01-29:
@chris-luangrath Didn't we decide that this was good to close?

vinhowe, 2021-04-29:
Fixed in https://github.com/BYU-PCCL/holodeck-engine/pull/216.

closed 2021-04-29 by vinhowe